baronbrew/TILTpi #3: persist logging configuration through reboot

bakerag1, 2018-03-28:
It seems to freeze up sometimes, reboot fixes it, but I have to reconfigure the cloud logging. It would be nice for it to just keep it.

noahbaron, 2018-03-28:
This issue should be fixed on the latest version, you can update it by going to the menu System->App Admin->Update App. Let me know if that fixes it.

bakerag1, 2018-03-28:
I updated to 2.1, set the beer name and cloud URL, logging worked. Then I disconnected and reconnected the power. The configuration is gone.

noahbaron, 2018-03-28:
Settings as well as the most recent data points are saved to the SD card every 15 seconds, so it's possible the setting wasn't saved if you restarted immediately after making the setting change. You can see what data has been saved to the SD card by going to http://tiltpi.local:1880/data/BLACK.json if you have a black Tilt. If you have another color, just change the color, for example a red Tilt would be http://tiltpi.local:1880/data/RED.json.

bakerag1, 2018-03-28:
I don't think I can walk that fast, but I'll try again and see. Thanks!

bakerag1, 2018-03-30:
It was still running since my last comment. I powered it off and on again. The UI showed everything but the custom cloud URL. I opened the json file and it seemed like everything was still there. I checked on brewstat.us and it is still receiving data.

So, I guess the only issue is that the URL doesn't show up in the UI after reboot. 

Thanks for your help and the cool app!

noahbaron, 2018-03-30:
You're welcome, glad you got it to work. I'll take a look at the UI issue you're seeing. Let me know if you see anything else that's strange.

closed 2018-03-30 by noahbaron