baronbrew/TILTpi #5: Logging does not record temperature in set unit

greyghostbrewery, 2018-11-09:
The log files only include the temperature in Fahrenheit even when the display unit has been changed to Celsius.

billsegall, 2018-12-22:
This is also the case on the main TiltPi display.

noahbaron, 2019-01-11:
I agree, it's an odd representation. It's the default coloring using the node-red UI node plugin. Something to investigate on the next version.

Regarding the display units vs. logged units. The assumption was that any CSV file would be used with a pre-formatted template in Excel so any unit conversions would be done there. It also prevents the numbers from going haywire should the units be changed mid-way through a brew. One work around would be to have a master log file with the native units and upon download convert the units based on the current selection.

billsegall, 2019-01-12:
As a representative of a country using the metric system (the exceptions being Myanmar, Liberia and the USA) I would obviously like to see Celsius being the default :-)

You might choose kelvin as an independent option but ultimately it's kinda futile when the tiltpi itself is designed for the US market and reports in farenheit and has a granularity in farenheit (.5 deg).

Another option would be to use the locale to make this choice in the log, or as you suggest to allow the logging units to be set. 

My issue was more in the UI and I'm happy with it now that I know how to make it appear in my units of choice. My apologies for hijacking this issue with that.