baronbrew/TILTpi #6: Changing hostname to FQDN loses flow

billsegall, 2018-12-22:
To duplicate, use raspi-config to change the hostname from "tiltpi" to tiltpi.your.domain" and reboot.

After getting past [chrome bug 367048 ](https://bugs.chromium.org/p/chromium/issues/detail?id=367048) travelling to [http://10.0.0.215:1880/ui](http://10.0.0.215:1880/ui presents a node-red no flows present screen.

Reverting the hostname change fixes it.

noahbaron, 2019-01-11:
I was aware of changing the hostname also required a change to the node-red flows file name. This is in the node-red documentation under [node red configuration](https://nodered.org/docs/configuration). I didn't know it was possible to use a global hostname. How does this work? Do routers automatically know how to handle this?

billsegall, 2019-01-12:
Setting hostnames using an FQDN is an old habit of mine. I used to deal with too many machines in too many organizations.

I think the issue relates more to **any** change of hostname which some people will want to do to make it fit their local network schemes etc. 

It is be  possible to make node-red immune from any  hostname change by explicitly setting flowFile in settings.js. This would be generous to do on the disk-image as it was difficult to diagnose what had gone wrong as a node-red and tiltpi newbie.