baronbrew/TILTpi #8: Enabling https requires the acl package for setfacl

billsegall, 2018-12-22:
The acl package should pribably be added to the dependencies and to the disk images

noahbaron, 2019-01-11:
Thanks for your comment. Is there a good reference you can point me to? Is this to setup node-red as an https server?

billsegall, 2019-01-12:
Yes, I was trying to set up node-red over https as I wanted to secure a tiltpi install so I might use it from outside my firewall. The error message was quite confusing and since I'd diagnosed it I thought it might be usefully done in the next release. 

Ideally the disk image would have https enabled by default but I should mention that I had more problems than this issue and did not get it working. I'll revisit at some point.

The disk image's ```/home/pi/.node-red/settings.js``` file is where the config gets done and you also need a self signed key (or perhaps letsencrypt might be better).

node-red doc is at [https://nodered.org/docs/configuration](https://nodered.org/docs/configuration)