baronbrew/TILTpi #9: global.json and <color>.json written to SD every 10 seconds

fgleason, 2019-01-11:
Could those files be disabled by default? They appear to be unnecessary SD writes. If they are required could you write to /tmp and use tmpfs for /tmp?

noahbaron, 2019-01-11:
Thanks for your feedback. Those writes are for restoring the system during a power outage or reboot. I understand there is a disadvantage to excessive writes to the SD card. If there is a more reliable memory location to write that can be accessed between reboots that would be ideal. If you have any references for tmpfs, let me know and will be happy to change.

fgleason, 2019-01-11:
Hmm that is a challenge.  Do you need the current values to restore after reboot or just the configuration information?  If it's just the configuration it really only needs to be written when it changes. Perhaps nice to have but not required to restart information could be disabled by the user in the GUI if they want to. Or maybe write less frequently.  It does not sound like tmpfs would work for this.  Here is a reference: [http://man7.org/linux/man-pages/man5/tmpfs.5.html](url)

noahbaron, 2019-01-11:
Just the configuration information, basically anything the user inputted like preferred units, calibration, beer name, cloud logging id, and custom cloud URLs. I could reduce the writes by only writing if the configuration data changes and not if the values like SG or temperature change, which would be a substantial reduction

fgleason, 2019-01-11:
That would be great. Thanks very much.