baronbrew/TILTpi #10: Tilt Pi not consistently uploading to Google Sheets

gifty74, 2019-02-14:


noahbaron, 2019-02-15:
We're aware of the issue and it's due to too many Tilt Pi's connected to the default Google Sheet cloud URL. One part of the solution is to create a new, separate cloud URL for each tilt color which can selected on the logging page.

gifty74, 2019-02-15:
Thanks Noah. So would a temp solution be to configure a different (less used) color on initial setup of the RPi?

noahbaron, 2019-02-15:
The temp solution right now is to copy our logging script and run it on a
an existing or new gmail/google account. We have instructions on how to do
this on our “App” page here:
https://tilthydrometer.com/pages/app

Check out “Option 2” under cloud logging apps.

noahbaron, 2019-03-10:
Good news! I was able to improve performance on this script by fine tuning where the locking mechanism was used to prevent two beers from being started at the same time and crossing paths.

gifty74, 2019-03-10:
Ok, I just saw this reply. Will try it today!