baronbrew/TILTpi #11: scanBeacon not working

hapklaar, 2019-02-23:
Followed your instructions on this gh main page to install the software in my RPi3B, however the scanBeacon part won't load. In the webpage GUI is get:

```
Flows stopped due to missing node types.

scanBeacon
```

sudo systemctl status nodered.service gives:
```
Feb 23 18:34:36 HapPi3 Node-RED[8342]: 23 Feb 18:34:36 - [info] Waiting for missing types to be registered:
Feb 23 18:34:36 HapPi3 Node-RED[8342]: 23 Feb 18:34:36 - [info]  - scanBeacon
```

It seems node-red-contrib-bleacon is not correctly installed or not working. Any idea what this can be?

antonievw, 2019-04-21:
Did you ever resolve this issue?  I’m experiencing the same problem.

noahbaron, 2019-04-21:
I only tested these instructions with Raspbian Stretch March 2018 (https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2018-03-14/). I recommend starting with this disk image and then updating to the latest version of Stretch once installed. I plan to replace the node for the next version of Tilt Pi so it can be installed on more systems, but for not its fairly restricted to this version of Raspbian.

TroyHomeBrew, 2019-06-12:
I can verify this is happening on the current version of Stretch.
The Bleacon package will not install properly with the commands in the directions. It fails with an error of
`WARN EACCES user "root" does not have permission to access the dev dir`
If you add --unsafe-perm to the install command it looks like it finishes but with tons of warnings.
Then if you finish the install directions you get the result that this post is about.

I have an always on Pi 3 running Stretch that's doing several other things that would be best to add onto and a second Pi running BrewPi but it's on Wheezy that I don't dare touch it because it works perfectly.
Maybe I'll pick up a Zero and give it a go with the provided image.

jgwehr, 2019-08-08:
After some fumbling through my pi, I've made some progress with node-red I thought I'd share. It's worth noting every few seconds I get errors, so this doesn't completely solve the issue....

> My primary conclusion is that Aioblescan seems to work with latest version of Stretch: _Stretch 9 downloaded on Aug 7 2019_

**Steps**
1. I followed these instructions (for Jessie) and had the same issue with the Bleacon package. 
1. _I would have preferred to uninstall node-red and even npm/n/node at this point but I'm not smart enough and didn't have any luck online._
1. I upgraded my pi from Jessie to Stretch. The only other thing I have on this pi is `pihole`.
1. Follow the instructions for the Aioblescan: https://github.com/baronbrew/TILTpi/tree/Aioblescan
1. Given I couldn't freshly install node-red, I had some fool's errands until I finally got the new flow to install. Again, I am **not** good with linux so, while I have the best intentions, running these commands may not be advised.
   1. `cd ~./node-red`
   1. `rm flows_raspberrypi.jason` (only do this if Tiltpi is your only flow)
   1. `sudo systemctl restart nodered` _This restarts the service and allows node-red to rebuild the flow file._
   1. Repeat the installation step 6: `wget -O /home/pi/flow.json https://raw.githubusercontent.com/baronbrew/TILTpi/Aioblescan/flow.json`
   1. Repeat the installation step 7: `curl -X POST http://localhost:1880/flows -H "Content-Type: application/json" -H "Node-RED-Deployment-Type: nodes" --data "@/home/pi/flow.json"` Node-red will pick this up without a restart
1.  Continue to your pi's URL on the :1880 port.
1. The /ui interface now appears to function well. I haven't been tried to connect the actual device, yet.

**Errors... to be continued**
Having gotten through this ordeal all of the node-red `nodes` are installed. However, every few seconds I get a few errors from at least 3 different nodes:
>TypeError: Cannot read property '1' of undefined
>SyntaxError: Unexpected token u in JSON at position 0

@TroyHomeBrew - I hope some of this helps. Probably just using that "new" install script will get you there :)

TroyHomeBrew, 2019-08-09:
Thank you for the work an the informative post, @jgwehr.

I ended up picking up a Pi 3B+ for a song after the 4 launched, used the provided image, and it's working great (well, close to great...my Tilt was reading about 12 points lower than my hydrometer out of the box and I still need to do a proper battery removal calibration).

It's slightly inconvenient to have another Pi running so maybe I'll give it another go with the new script next time I have a few hours to tinker with it again.