baronbrew/TILTpi #12: TiltPi not consistently uploading data to BrewFather

SmokeyRider, 2019-03-02:
I'm seeing missing data points on my BrewFather log.

I noticed an issue uploading to Google Sheets but that seems to be related to overloading a single Google Cloud URL.  I would assume that this is not the case for uploading to BrewFather since they have separate URLs (e.g. http://log.brewfather.net/tilt?id=xxxxxxx)

noahbaron, 2019-03-10:
Thanks for your comment. I was able to improve efficiency of the Google Cloud URL and that should be fixed now. I know that brewfather doesn't allow posting more than every 15 minutes, could this be the issue? Also can you confirm that WiFi connection is good for your situation?

gifty74, 2019-03-10:
So Noah, I assume it’s still best to create a custom Google Sheets URL and you are saying that we should not have as many missing data points now? I do use a custom url, but was still missing some data points throughout a brew session. Sometimes up to 3 in a row.

noahbaron, 2019-03-10:
The custom Google Sheets URL (option 2) is more difficult to setup so at this point, with this latest fix/performance improvement, I recommend using the default. That said, option 2 is still really good and I have updated the template with the same code. I also have the code on github since it runs independent of the Tilt Pi.  Google provides a dashboard to the script health (in the script editor go to menu View->Stackdriver Error Reporting) and the errors appear have stopped completely. Previously I was seeing thousands of errors/missed data points. It appears to be running faster too on my test Tilt Pi.

If data points are still being missed, it's most likely due to the WiFi connection. The next Tilt Pi update I'll look at stopping the Bluetooth scanning temporarily while logging to the cloud. Bluetooth and WiFi share the same antenna and have seen performance issues when both are highly active. I will also have the Tilt Pi attempt to log again if a success message wasn't received.

gifty74, 2019-07-01:
Hey Noah, still having issues getting consistent uploads to my google sheet. Using the custom URL option (after making a copy of your sheet, registering it as an app with sheets, etc). Any ideas? It looks like you last thought this was fixed. I'm also stumped as to why more people aren't having the same issues as me. No problem with proximity of the Tilt, it has a good strong connection to it, and within 10 feet of my wifi router.

noahbaron, 2019-07-01:
Thanks for your follow up. I checked the error report on the new script and
I’m not seeing errors there. Are you seeing the Tilt Pi ever disconnect
when viewing the dashboard from a phone or laptop? Do you see any error
messages when manually logging several data points at a time?

gifty74, 2019-07-01:
No errors I can find. The internal log on the Pi is rock solid, never missing a data point. Logging to the cloud though is where it's hit or miss. Link here: https://docs.google.com/spreadsheets/d/1QroR1jTh_i5Hyzei5d_S0oeWjnvkLIjf9gV9Ti2Ldgg/edit#gid=734290882

Maybe i'm doing something wrong in the publishing of the google sheet app? I'm following the instructions as far as I can tell.

noahbaron, 2019-07-01:
Thanks for sharing the Sheet. I see the skips in timepoints. One way to see
if there were server errors with your Google Sheets script is to got to the
Script Editor view (When looking at the list of beers go to Tools->Script
Editor...) and then go to the menu View->Stackdriver Error Reporting. If
you have no errors you will see a message to setup error reporting as shown
below:
[image: image.png]
if you do have errors you will see more detail about when the errors
occurred and how many as shown below screenshot.
[image: image.png]

lbussy, 2019-07-01:
It’s possible that it’s just the RPi wireless being flakey.  If it’s a network timeout (and you are on wireless) you can use [this standalone daemon](https://github.com/lbussy/rpi-wifi-checker) to help.

noahbaron, 2019-07-01:
Thanks @Ibussy, Will test this script as well, maybe integrate a link to
the log output into the settings page on Tilt Pi.

One thing to increase stability is to stop BT scanning temporarily while
attempting to log data of shared WiFi/BT antenna.

gifty74, 2019-07-01:
If the bluetooth/wifi interference is a known problem, I would think the code would want to disable the bluetooth for 10 sec while the Pi uploads the data to the google sheet. Maybe this is easier said than done...

gifty74, 2019-07-01:
don't see any errors...

[image]

noahbaron, 2019-07-01:
In my test RPi setups I haven't seen it as a problem. A few Apple devices clearly have this problem (for example my iPhone Xr), and for these I do turn off BT scanning while logging, which fixes the problem. Normally the operating system should deal with this more effectively, but it's not alway the case.

It appears you're checking the script editor on the actual brew log. You need to check the script editor on the your Sheet where the script is actually attached, where you got the cloud URL for the Tilt Pi. There should be a tab called "Beers" with a list of all the beers you have brewed. From there, you can check for errors.

gifty74, 2019-07-01:
Yeah, none in Script sheet either.

gifty74, 2019-07-02:
Since installing that wifi 'keep alive' script last night it has not missed one log point to the google sheet. not sure what exactly that did, but might want to look at including that in TiltPi for those that use wifi (which I would think would be the majority?).

lbussy, 2019-07-02:
Glad it worked for you.  Noah, you can grab it with my blessings.  

The RPi for some reason doesn’t like to stay connected consistently.  Part of it is power management, part of it is sunspots I guess.

gifty74, 2019-07-02:
Spoke too soon. Today throughout the day missed a bunch of data points on the Sheet.

gifty74, 2019-07-02:
I have an old Kindle Fire that I've keep running nearby that NEVER misses a beat (data point). Odd that a RPi, which is a fully capable PC, does not reliably publish the data when given very specific instructions via code. You'd think that a computer would be able to handle this function, and do it repeatedly without issue. Would much rather use the RPi, but the Fire is much more reliable.

lbussy, 2019-07-02:
I'd stop short of calling RPi a fully capable PC.  The Fire is probably a better platform all in all - just not as open.

No you'll never make the RPi rock-solid, but the script should help.

noahbaron, 2019-07-03:
Another option I've seen to improve WiFi connection is giving the raspberry
pi a static IP address in your router settings. This improved WiFi for some
Apple devices I've used in the past.