baronbrew/TILTpi #13: Unstable connection between RBPi3 and Black Tilt

pottati, 2019-05-25:
Hi, 

I've installed Tiltpi HDMI image everything works both google sheets and brewfather. But frequently I can't see the reading on the ui. it's like connection appears and disappear randomly. have you ever expirienced issues like this?

KR
[image]

pottati, 2019-05-25:
It seems like color is not properly read. when it works black appear but now it doesn't work. other screenshot. Hope that they will help. KR
[image]
[image]
[image]

noahbaron, 2019-05-25:
What you're seeing is consistent with the Tilt being out of range or the signal being blocked. It's best to have the raspberry pi clear of any unnecessary wires, metal, or any electrical devices. When you see the Tilt Pi connect, check the RSSI (signal strength) values on the bottom of the readout. Consistent numbers between -90 dBm to -105 dBm suggest a range issue.

pottati, 2019-05-27:
RPi is really close to the fermenter  less than 1 meter. The tilt is in a Grainfather conical fermenter.

noahbaron, 2019-05-27:
With the Grainfather’s double stainless walls, I would position the RPI on
top of the fermenter. Which RPI do you have? I notice the RPI3 is better
than RPI 0w in regards to reception.

pottati, 2019-05-28:
RPi3. And Yes signal is not that good -80 to -90. Let me say that is a borderline situation.

noahbaron, 2019-05-29:
Our solution for this type of problem is to use a repeater. We developed a script that runs on a low energy bluetooth module. The module is battery powered and can scan a specified color of Tilt and relay data to an RPI and phone. We have a blog article about it here: https://tilthydrometer.com/blogs/news/extending-range-in-stainless-fermentors-with-the-tilt-repeater-1