baronbrew/TILTpi #14: Docker Image

seanarnold, 2019-10-05:
It would be great if there was a docker image where we could spin up TiltPi with minimal config. This should remove errors of node versions, raspbian being updated etc.

Is this something that has started anywhere? Otherwise I'd be keen to help.

noahbaron, 2019-10-05:
Hi, thanks for the suggestion. I’m not aware of any docker image existing
for Tilt Pi, sounds like a convenient option. Thanks for offering to help!
Let me know if you need any details from me to get this started.

Cheers,

seanarnold, 2019-10-07:
I've been trying to get this working using the instructions in the README but I'm getting the following errors when loading the UI

```
Flows stopped due to missing node types.

ui_group
ui_tab
ui_base
ui_link
scanBeacon
ui_text_input
ui_template
ui_dropdown
ui_switch
ui_button
ui_toast
ui_slider
ui_text

```

Any tips?

seanarnold, 2019-10-07:
Just found the updated instructions here: https://tilthydrometer.com/blogs/news/install-tilt-pi-on-raspbian-buster-compatible-with-all-rpi-models-including-rpi-4

Seems to work

DevMattM, 2019-10-07:
How did the Docker image turn out?

noahbaron, 2019-10-07:
Great to hear the new instructions are working. Always a challenge as
different components are updated.

gifty74, 2019-10-08:
I have a Pi 3B+ and the latest tiltPi build with Buster (hdmi version) will not run for me for more than a few hours without freezing. Tried a different SD card, updating/upgrading, new power supply...everything. Looking in the log there are a ton of Bluetooth errors, one ever few seconds. I think it might be overloading the Pi??

noahbaron, 2019-10-08:
That was a problem with the first release of the Tilt Pi image, did you try
updating the app in the system menu? The update manages the errors by
killing and restarting the process after an error is outputted from
Aioblescan. I haven’t been able to determine the root cause of the errors.

gifty74, 2019-10-08:
> That was a problem with the first release of the Tilt Pi image, did you try updating the app in the system menu? The update manages the errors by killing and restarting the process after an error is outputted from Aioblescan. I haven’t been able to determine the root cause of the errors.
> […](#)

I did. Updating in the app doesn’t seem to do anything. The button flashes briefly but nothing else. I also restarted and updated everything else I could. Here’s a screen shot of the ongoing errors, one or two a second. 
[image]

noahbaron, 2019-10-08:
I'm seeing those errors on my end as well. I tried searching for the error
online and it appears it could be related to the bluetooth driver on this
version of Raspbian. Regardless, Tilt Pi v2.3 should handle these without
overloading. (can you confirm you are using Tilt Pi v2.3 you see in System
menu?)

I've been running Tilt Pi v2.3 server version for several months now
without a reboot. I'll start testing the HDMI/GUI version since there might
be a difference.

gifty74, 2019-10-09:
Yes, version 2.3, HDMI version. It appears it’s an issue with the pi3 and 3B. I found a thread that says to lower the baud rate of the Bluetooth controller in usr/bin/btart, but I can’t figure out how to get the proper permissions to modify the one parameter they say to change. Permissions are for root, not pi.

gifty74, 2019-10-09:
Ok, lowered the rate in the btuart file and that seems to have fixed it. Just not sure if the range is now affected. Will try running the server near my tilt tonight.

noahbaron, 2019-10-09:
That's great to hear. What commands did you use?

gifty74, 2019-10-13:
Ok, so after testing it appears that lowering the bt rate eliminates the errors I was getting, and has stabilized my Pi3B running TiltPi v2.3. Over the past 24 hrs it has not missed one data point, and seems to have a much more accurate recording time. Prior, I was missing random points, and often times it would record a data point and the 15 min interval would start to drift. If it started recording at 8:01:05 for example, several hours in it would be at 12:02:30 for example. The interval would slip, presumably from the mass amount of errors jamming the system and delaying the upload? Not really sure on that, but this tweak has totally stabilized my Pi3B, so far.

noahbaron, 2019-10-13:
That's really great news! I found and edited the btuart file you were
mentioning and am trying this as well. Should probably add this to the SD
card image to improve stability.

gifty74, 2019-10-13:
Yeah, test it out and let me know!

lbussy, 2019-11-23:
> I found and edited the btuart file you were mentioning and am trying this as well. Should probably add this to the SD card image to improve stability.

I assume you ended up editing this?

https://github.com/RPi-Distro/pi-bluetooth/blob/cbdbcb66bcc5b9af05f1a9fffe2254c872bb0ace/usr/bin/btuart#L17-L26

What value did you end up using?

I checked the current version of TiltPi but I guess you've not had a chance to update yet.

gifty74, 2019-11-24:
Yes, that file. Drop the 921600 to 460800. Make sure to look afterward at the logs. And I agree, the image should be updated, or have a special version for the Pi3B with the lowered bt rate.

lbussy, 2019-11-24:
Thank you sir.  I’ll give that a go.

lbussy, 2019-11-24:
I have a Pi 4.  It looks as if the baud rate would be set at 3000000 which seems pretty high to me.  I'll have to do some more testing.  Maybe the Pi 4 can handle the higher rate?

lbussy, 2019-11-25:
Ran overnight on a Pi4 at 3000000 and no errors in syslog.  I'll try next with one of my 3's.  Of course, I'm just recreating your findings, but I thought sharing my Pi 4 results would be helpful.

noahbaron, 2019-11-25:
I've been trying to figure out the least painful way to fix this. Most likely I will add a button in the System menu that will generate the modified "btuart" text file and replace the current one. The specific part of the file that needs to be changed is line 22 in /usr/bin/btuart I've changed the baud rate to 115200 since I was seeing errors even at 460800.

gifty74, 2019-11-25:
You are correct, I also dropped it to 115200. The Pi4 should handle the errors much better than the 3B+, which is why @Ibussy is probably not seeing the errors. But yeah, a button would be nice.

lbussy, 2019-11-25:
So what I did for the BrewPi Remix script (at least this is what I have right now) is:

``` bash
do_aioblescan() {
    # Install aioblescan
    local blerepo device fast safe file
    echo -e "\nInstalling BLEacon support via aioblescan."
    blerepo="https://github.com/baronbrew/aioblescan.git"
    file="/usr/bin/btuart"
    fast="\$HCIATTACH \/dev\/serial1 bcm43xx 921600 noflow - \$BDADDR"
    safe="\$HCIATTACH \/dev\/serial1 bcm43xx 460800 noflow - \$BDADDR"
    rm -fr "$HOMEPATH/aioblescan"
    git clone "$blerepo" "$HOMEPATH/aioblescan"
    (cd "$HOMEPATH/aioblescan" || exit; python3 setup.py install)
    rm -fr "$HOMEPATH/aioblescan"
    # Slow down uart speeds on < Pi4
    if [ -f "$file" ]; then
        sed -i "s/$fast/$safe/g" "$file"
    fi
    device=$(hciconfig | grep "hci" | grep "UART" | tr -s ' ' | cut -d":" -f1)
    if [ -z "$device" ]; then
        if grep -vq "Pi 4" /proc/device-tree/model; then
            stty -F /dev/serial1 460800
        fi
    fi
}
```

Note that $HOMEPATH is set previously so you'll want to address that.  At least to my way of thinking this will only address baud rates for Bluetooth devices connected via the UART, and then only that BAUD rate which was troublesome previously.  Then for good measure, it resets the baud rate of the existing serial port for "right now" work.

Minimally tested, but you get the idea.

noahbaron, 2019-11-26:
Thanks! Stability fix is now available as a button in version Tilt Pi v.2.4. Just tap the "Update App (Flow)" and you'll see a button appear that will allow you to make the baud rate changes. I also added a button to restore in case some other fix ever becomes available.

If the Docker Image can avoid problems like these, I'm all for it. Not sure how though, I would think the docker image would have the same problem.

dirtboy12, 2020-04-23:
I almost have this working in Docker for RPI4, but I am running in to a problem with it keeping my settings.  When I restart the system it reverts back default settings.  I've tried to set the default cloud URL in the settings files that I can find, but I am not having much success.  Is there a specific place I can set all my settings and have it use that on startup?

noahbaron, 2020-04-23:
That's really cool. Will this be available to download?

Regarding your question, the system or global settings are kept in a file
called "global.json" in the location "/home/pi/global.json" There are also
color-specific settings files stored in the same folder. For example
settings for a black tilt are stored in "/home/pi/BLACK.json" (note that
the color is spelled out in all caps). When the system starts up, these
text files are turned into a JSON object and used to restore settings.

dirtboy12, 2020-04-24:
I hope so.  I have to do some testing.  It would be my first distributed image so I want to be careful. 
 So far I have had to do this:

- Edit update-nodejs-and-nodered script to remove prompts for Dockerfile build compatibility (that file hasn't changed in the last 9 months, has it?)

- Attach a volume for /home to write configs (I hope to not have to do this if I can get defaults to work for me)

- Add pi to sudoers for a short amount of time to complete node-red installs

- Find a time when my tilt is not in the fermenter so I can have something to read from

It has worked in the past, but it would require me to login via console to launch processes.  Trying to fix that.  I also need to test to see what the effect is of replacing the /home directories with a mounted volume since the node-red installs write so much to the ~/.node-red folder of the user as part of the install process.  If I distribute an image with node-red installed and the user mounts their own writeable volume over that /home folder, will node-red work, or does the user need to run the installs again?  We will see.

dirtboy12, 2020-04-29:
Anybody want to test?

https://hub.docker.com/repository/docker/dirtboy12/ubutilt

Raspberry Pi 4 only and you need to have --net=host.

I tested using Raspbian Docker.  See how it works.  Any suggestions for decreasing the size are welcome.

noahbaron, 2020-05-01:
Thanks, will give it a try. I’ve never run docker so might take time to get
up to speed.

jgwehr, 2023-05-02:
> I've been trying to get this working using the instructions in the README but I'm getting the following errors when loading the UI
>
> ```
> Flows stopped due to missing node types.
>
> ui_group
> ui_tab [...]

I'm trying to tackle this myself, now. Didn't really have any luck with the ubutilt image but I might try later.

I did find how to progress past this initial issue: all of these "missing" elements should be installed prior to importing the TiltPi  flows using this: [node-red-dashboard](https://flows.nodered.org/node/node-red-dashboard)

The major note for @noahbaron brew is the disclaimer on this extension, though:

> This project is based on Angular v1 - As that is now no longer maintained, this project should be considered to be on "life support"

It does feel like the overall implementation for this needs a revisit. "Life Support" isn't great for a device still being sold; getting it to work on Docker would at least stabilize this as other pieces move forward.
I'll try to see if I can get it running with the node-red docker image, but I have a feeling the bluetooth will be a pain.

noahbaron, 2023-05-02:
Joe,

Thanks for your feedback. I have update the instructions on our website
here:
https://tilthydrometer.com/blogs/news/install-tilt-pi-on-raspbian-buster-compatible-with-all-rpi-models-including-rpi-4

I agree, the version of node-red UI and node-red core is ancient in
software terms and needs to be ported to a new version. I have also noticed
the latest version of Raspberry Pi OS  (February 2023) doesn't support
installation, however the September 2022 version does work. I have seen
examples of docker images using Bluetooth, notably this one:
https://github.com/balena-labs-projects/bluetooth

jgwehr, 2023-05-11:
@noahbaron I have some promising progress! This may not be the best place to ask questions or bother you, let me know.

So far I have node-red being prebuilt with any needed pallette/nodes. I also have successfully polled BT devices!

This leaves the TiltPi UI reachable - but "stuck", presumably because aioblescan isn't available. I've been testing from the ground up with `node-red-contrib-noble-bluetooth`. I can successfully get nearby devices, but **I'm not able to determine which is my TiltPi**. I gather from your flow you have access to and filter on static UUIDs you must set for the devices - but this node component doesn't provide that:

```json
{"peripheral":"27f43cf96be2","address":"27:f4:3c:f9:6b:e2","rssi":-71,"connectable":false,"manufacturerData":[76,0,9,6,3,81,192,168,1,34],"services":[],"_msgid":"fceb43e331ca8e33"}
```

Any idea how to identify the Tilts or am I on a goose chase with this component?

noahbaron, 2023-05-11:
@jgwehr 

Thats great news, glad to hear you made some progress. I recommend using the reference here to scan for Tilts: [https://kvurd.com/blog/tilt-hydrometer-ibeacon-data-format/](https://kvurd.com/blog/tilt-hydrometer-ibeacon-data-format/)

The Tilts are non-connectable and only use the manufacturer data per iBeacon specification. The output of Aioblescan looks like this:

{"uuid": "a495bb10c5b14b44b5121370f02d74de", "major": 30, "minor": 1065, "tx_power": -59, "rssi": -74, "mac": "fd:35:71:48:27:0d"}

davismariotti, 2023-07-09:
I was interested in creating a docker image for myself to use with a bluetooth dongle, seems to be working pretty well, available here: https://github.com/davismariotti/tilt-docker

Many of the flows were modified to remove Raspberry Pi specific nodes (such as reboot/shutdown/etc).

jgwehr, 2023-07-10:
@davismariotti Well done! I had to change a few things to get it to work on a Raspberry pi, but it is working, thank you!

Specifically, I both the node and python versions were causing issues for ARM.
`FROM node:latest`
`RUN apt-get install -y bluez bluez-tools bluez-hcidump bluetooth unzip wget usbutils git python3.11 tzdata`

I also noticed the image size (~1gb) is 2x what my build off `nodered/node-red` was (~435mb). I might try to play with the dockerfile a bit more to see if it can run as well off nodered instead of node. I was totally stuck with the bluetooth and you've got some great hints for me!

thorrak, 2023-07-10:
I actually have a similar docker image that I put together as part of a set of upgrades to Fermentrack that is available [Here](https://github.com/thorrak/tiltbridge_junior) 

It’s not (currently) as feature-rich as TiltPi, but the image size clocks in at about [25 MB](https://hub.docker.com/r/jdbeeler/tiltbridge-junior/tags)

davismariotti, 2023-07-10:
@jgwehr Fair point on the size. I modified it to use alpine as a base image, image size is now 277MB uncompressed. Pushed v1.1 to docker hub which has multi-arch builds now, hopefully will work on Raspberry Pi/ARM as well.

dirtboy12, 2023-07-10:
I'm glad you guys picked this up. I was considering updating the image but
I lost the original scripts I had used.

Plus it looks like you guys have more experience with docker than I do and
more time.

Cheers!