baronbrew/TILTpi #15: Returning Battery Life Consistently

lbussy, 2019-11-23:
@noahbaron hopefully this is the right place to ask this question.  I did not see a better place.

I think I'm missing something simple here.  I see the battery (tx_power) on my beacon vacillating between 0x20 (32 weeks) and 0xC5 (-59 signed or 197 unsigned).  I'm sure I'm missing something really simple here but apparently I'm being dense today.

Help a guy out?

noahbaron, 2019-11-25:
happy to help! The numbers you're seeing are expected. It's not documented anywhere, but as you noted, tx_power portion of the iBeacon data packet is in weeks since battery installation with the exception of -59 (197 unsigned). This number alternates every 5 seconds so that iOS can read the RSSI succesfully. If iOS gets an "out of range" number for tx_power (like 1 or 2) it will report the RSSI as "unknown". So we chose to alternate to -59 because that's a default value for many iBeacons and iOS will be properly report the RSSI. If building software that displays battery status, we recommending filtering out the 197 values. We'll likely do that soon where the data is posted as 197 when logging locally to a file.

lbussy, 2019-11-25:
Oh I get ya now, thanks for clearing that up.  I over-thought it. :)

So are you saying that the RSSI will *always* report -59 when it "allows" iOS to read RSSI, or will it be some other negative number?

noahbaron, 2019-11-26:
It would be some other negative number. iOS reports RSSI normally (usually
ranging from -25 to -100) if the tx_power is set to -59. I’m sure other
numbers nearby would work too but -59 was a default on the Bean board we
first used to build Tilt, so all the Bean based Tilts transmit “197” so
easy to filter this number out.

lbussy, 2019-11-26:
Sorry I got myself crossed up there between TX_PWR and RSSI.  :)  I get it now.  Thanks Noah!

closed 2019-11-26 by lbussy

zrrbite, 2022-05-22:
Is this issue for V3 of the Tilt? From what i'm reading here, and looking at the packets from the one V3 I own, it does seem like TX_PWR / RSSI values are related to battery life. Did this change for newer versions of the Tilt? 

Thank you.
/Martin

noahbaron, 2022-05-23:
Martin,

Thanks for your question. There haven't been any changes regarding the
TX_PWR values discussed here. The TX_PWR byte still shows the battery age
in weeks alternating every 5 seconds with a value of -59 (197 unsigned).
RSSI is different from the TX_PWR byte and it changes depending on how
close or far (or how much interference there is) with the Tilt. Hope that
answers your question.

Regards,

Noah

zrrbite, 2022-05-24:
Hi Noah,

Thanks for getting back to me. This Tilt was purchased relatively recently, so it should have the battery update. I'll have another look at it, maybe I'm off by one :)

Best regards,
Martin

lbussy, 2022-05-24:
@zrrbite please feel free to compare your work to kine here:

https://github.com/brewpi-remix/brewpi-script-rmx/blob/main/Tilt.py

It may help you find the difference.

zrrbite, 2022-05-28:
Thank you @lbussy !