baronbrew/TILTpi #22: Logging only 1 pt now with 2.6 update

gifty74, 2020-04-12:
Updated Flow to the newer 2.6 and now my Pi will only log the first data point and none after that. Tried restarting, changing the name, etc, everything I could think to get it back on track with logging. Here's the sheet...

https://docs.google.com/spreadsheets/d/1oB6CNafX0irrVaozvBqqAOiYDv6_cS6rDVlE5jWWoPA/edit#gid=734290882

noahbaron, 2020-04-12:
Thanks for letting me know. What’s showing in the beer name field? Are you
able to manually log data?

*QuarantineSaison2,25634*

gifty74, 2020-04-12:
Now getting the confirmation email every 15 mins. The email saying u have edit rights to the google sheet. Odd.

gifty74, 2020-04-12:
Any way to get back to v2.5?

noahbaron, 2020-04-12:
Sure, enter the following two commands in the command line / terminal app.
The command will download the older version on github (v.2.5) and the
second command will deploy the the downloaded file. Let me know if that
fixes the issue. Didn't think there were any changes affecting this aspect
of the app.

wget -O /home/pi/flow.json
https://raw.githubusercontent.com/baronbrew/TILTpi/4dca392c4d0f063731bcdd00df6c29313d939a96/flow.json

curl -X POST http://localhost:1880/flows -H "Content-Type:
application/json" -H "Node-RED-Deployment-Type: nodes" --data
"@/home/pi/flow.json"

gifty74, 2020-04-13:
The curl command not doing anything it seems. Rebooted and still on v2.6.

noahbaron, 2020-04-13:
Hmm, I actually tested this in one of my machines. Did you see the download
complete successfully when doing the first command? Maybe it’s a WiFi issue?

gifty74, 2020-04-13:
Yeah, the download works. Just when I run the curl command it goes down to the next line and there’s no message or confirmation or anything. Is that the way yours worked? 

-Steve

noahbaron, 2020-04-13:
Yeah, there’s no confirmation message after the curl command.

gifty74, 2020-04-13:
Ok. So after trying several things ended up going back in and loading the latest version of Buster and manually installing node red and the flow.json. All back and working now, on v2.6 (guess it grabbed the latest version). Will test stability on this beer over the next 48 hrs.

noahbaron, 2020-04-13:
Thanks for the update. Let me know if you have any other trouble with it.
The purpose of v.2.6 was to make it easier to set a custom source for
updating the app as well as check if there's an updated version available.
Shouldn't have any affect on cloud logging.