baronbrew/TILTpi #24: Feature Request: Set individual logging intervals per URL

Romfrosk, 2020-06-15:
Hi. 

I'd love to be able to log at different intervals to URLs.

BrewSpy can handle every 5m
Brewfather can handle every 15m
Google can probably handle every minute if not more often.

noahbaron, 2020-06-19:
Hi, I agree, that would be a great feature. We will definitely add it to
our roadmap. I assume that setting the interval to 5 min will cause a
problem if trying to use Brewfather?

Romfrosk, 2020-06-19:
Yes. They do mention that 15m is the lowest value.

I will test to set it at 5 minutes and see if it just skips the values coming in too quick, or if it also blocks those coming in every 15m.

Can't test until my beer has been bottled though.

tmack8001, 2021-01-26:
I believe from when I was looking at reworking the custom urls in nodered that yes BrewFather does ignore anything that comes in sooner than the 15m window, not sure of the specifics off the top of my head.

Overall this is a much welcomed addition. I'll look to see if I can resurrect my flow that had something similar to this, but missing edge case testing.

tmack8001, 2021-01-26:
This is what the UI looked like on my branch (not complete). It would store properties in a map data structure for each of the "cloud logging providers" created (frequency, url, enabled and name) I think were all specific to the provider being updated.

[image]

tmack8001, 2021-01-26:
For instance here I have brewfather (not named) and google sheets (though have the same time interval set for both right now).
[image]

When I built this I also thought we should make it so the user can't "delete" the google sheets entry, but rather just disable it thus the "enabled" toggle and an uneditable Default Google Sheets entry.

[image]

Not sure when I'll have time to come back to this and learn more about NodeRED to make sure I catch all the edge cases that might happen, but who knows might try to finish this off.