baronbrew/TILTpi #25: [Question] Can we rename tilt device from a color?

MattL-NZ, 2020-07-14:
We have multiple tilts (approx 20) and some are the same color.
Is there a way we can easily distinguish the different colors and what beers they are in by giving the Tilt device a different name. (e.g. Tilt purple 1, Tilt purple 2). 

Is this possible?

Thanks.

noahbaron, 2020-07-14:
The Tilt Pi is capable of reading the unique mac address transmitted by
each Tilt so in that way you could distinguish between Tilts of the same
color. However, Tilt Pi has not yet been set up to handle that
functionality yet.

MattL-NZ, 2020-07-14:
Thanks Noah. 
So are you saying we can actually connect TiltPi to more than 8 devices but we can only view 8 within the UI?

If the above is correct, would I be able to get the data from more than 8 devices (approx. 20) through Tilt Pi into Google Sheets?

noahbaron, 2020-07-14:
No, it's only possible with additional development work

MattL-NZ, 2020-07-14:
Ok, makes sense. 
I don't suppose you have an ETA on when something like this could be developed?

noahbaron, 2020-07-14:
It would be faster to add a "white list" feature that would filter for a
specific mac address. In this case you would use two raspberry pi's which
might make more sense since the range could be better optimized.

wiltdavi, 2020-07-15:
Could you already do what is being requested with 2 raspberry pi's?  There is the ability to Filter Tilt by Signal Strength.  By using this I assume you would be able to get the 2 raspberry pi's to only read a specific set of tilts assuming they are grouped properly.  This would perform the same action as the "white list".  From there you should be able to set each color associated with each raspberry pi to a different beer name and effectively log 2 tilts of the same color at the same time.

Note: This does require 2 raspberry pi's and the duplicate colors must be separated from each other so that they can be filtered by signal strength.

noahbaron, 2020-07-15:
That’s correct, this feature is already available and would work well if
it’s possible to use two or more raspberry pi’s that are separated into
different areas or rooms. We actually use this setup for calibrating Tilts
and it works, however there’s an occasional stray RSSI that gets through.

MattL-NZ, 2020-07-26:
Thanks for your suggestions guys. That is possibly what we will go with.
One last question however; Can we change/update the UUID of each of our devices? That way we could identify the differences between two tilts that are the same colour. Noah is this possible? Would be great if it was!

Thanks
Matt

noahbaron, 2020-07-27:
The UUID is actually what encodes the color. Unfortunately, there's no
interface for changing it. Definitely a good idea and something to consider
in the future. That said, I have made some progress with Tilt Pi using the
unique mac address to separate out devices of the same color. Will work on
it this week and let you know when there's a beta for testing.

MattL-NZ, 2020-07-27:
Thanks @noahbaron.

Let me know as soon as I could start testing the beta version. Hoping to use this before the end of next week if possible.

myoung34, 2020-07-28:
@noahbaron I'd love to see if we can come up with a collaboration if you're interested.

I maintain https://github.com/myoung34/tilty and currently have plans to build a much more user-friendly UI similar to tiltpi

noahbaron, 2020-07-28:
You can start testing now. Click "Update App (flow)" in the System menu.
This will add a text field UI component below the Update App button so you
have an option to test a beta version of Tilt Pi by typing "tiltpi-dev"
into the field. Note that once you do this the only way to go back is to
type "Aioblescan". "Aioblescan" is the default you're currently using. Not
sure how long I'll keep this feature, but it's the simplest way I could
think of to beta test. Let me know your feedback. So far my testing shows
that I can have up to 20 Tilts using the same Tilt Pi at once.

MattL-NZ, 2020-07-28:
Excellent, thanks @noahbaron. I will test this tonight NZT and will report back this time tomorrow.

MattL-NZ, 2020-07-28:
@noahbaron Is there a way to identify the differences between the two tilts with the same colour? Is this done by Mac address or can we change the name of each of them?

noahbaron, 2020-07-28:
@myoung34 A collaboration would be great. As you know Tilt Pi depends on
the dashboard node-red plugin and although easy to use has significant
limitations and requires some workarounds here and there. Feel free to
email me directly to follow up.

noahbaron, 2020-07-28:
The mac address is now appended to the color. You can still set the beer
name. So you could add an additional identifier to the beer name so that if
you had a grid of fermenters all with the same beer you could call the beer
A1 IPA, A2 IPA, etc and each would have a unique color plus mac ID
associated with it.

MattL-NZ, 2020-07-28:
Thanks. 
I guess I should set these up one at a time so I know which tilt is going into which fermenter. 
I would hate to name for example two black tilts A1 IPA and A2 IPA and forget which is which and put the wrong one in the wrong fermenter. Will report back with my testing.

noahbaron, 2020-07-28:
Agreed. It appears the barcode on the Bluetooth chip doesn't match the mac
address numbers so not much help there. If we need to keep track of an
individual Tilt, we'll set it on a sheet of paper with a label on it.

MattL-NZ, 2020-07-29:
@noahbaron Trying to follow your instructions but I cannot seem to make the text box appear when I click on the "Update App (flow)" button in the System menu. Do I need to completely re-install it on my Raspberry Pi?

noahbaron, 2020-07-29:
Which version of the Tilt Pi app are you using? It needs to be version 2.5
or 2.6. The original Tilt Pi versions can not be upgraded. You may need to
download a new image file

MattL-NZ, 2020-07-29:
Using an older version. 2.1 - just going through the update now. Will report back.

MattL-NZ, 2020-07-29:
@noahbaron working great! I can now distinguish between two tilts that have the same color. Great work. 
[image]

MattL-NZ, 2020-07-29:
Any plans on increasing the limit from 20 to more? The project we are working on will need to monitor 22-25 or so Tilts. Not urgent however as we could just use two Raspberry Pi's to do this. But appreciate your work @noahbaron.

noahbaron, 2020-07-29:
Thanks for the feedback, good to know you got it working! It would be
simple enough to increase to 25. Unfortunately it has to be a fixed number
since I need to set up a pre-fixed number of empty panels that the display
goes to. 25 seems reasonable. Let me know if you see any other issues come
up.

MattL-NZ, 2020-07-30:
Legend. Thanks @noahbaron. I can see the 25 squares now so will continue testing.

MattL-NZ, 2020-08-03:
Hey @noahbaron I have just installed Tilt-Pi on a brand new Raspberry Pi and can see only 20 squares. Is there anything I need to do to get the 25 squares/pi's showing instead of just 20?

noahbaron, 2020-08-04:
@MattL-NZ, it was always 20 squares, it's getting hard to count! Just finished adding 5 more squares so you should now be able to get 25 Tilts going at once. I haven't tested it with all 25 Tilts yet, let me know if you see any issues with logging or anything else.

MattL-NZ, 2020-08-04:
Oh really, I counted 25 a week ago but that could be a case of "too many beers that evening".
Thanks for adding 25 tilts.
I tested with 20 last night sitting in the same bucket of warm water and I could see the data for all of them. There was a few instances where they all disappeared for a second then after about 1 minute they all started coming back again.

noahbaron, 2020-08-04:
No problem, thanks for testing. What Raspberry Pi are you using? How close
is it to the bucket.. If you are seeing all disappear it means the Tilt
hasn't been seen for 2 minutes will cause everything to reshuffle.

MattL-NZ, 2020-08-05:
We are using the newest Raspberry Pi 4 Model B.
It is around 2 meters away from the bucket. 

Yeah the reshuffling looks to happen when more tilts get added to the bucket.
Example: there are 3 black tilts and 1 yellow tilt and 3 blue tilts in the bucket. If i go and add another yellow tilt, it reshuffles everything around so the two yellow tilts are now next to each other. Takes around 30seconds - 1minute to do this.

noahbaron, 2020-08-06:
Ok, good to know. It doesn't sound like there are any unexpected disconnects. The reshuffling occurs when a new Tilt is seen since they are all sorted alphabetically so colors are in groups and blank squares at the end.

MattL-NZ, 2020-08-10:
@noahbaron I am happy to close this issue/question. Everything looks to be working perfectly from our side of things. We're running around 22 tilts simultaneously. Will you be moving these adjustments into the production version of Tilt Pi or should we just continue to use the beta version?

noahbaron, 2020-08-10:
Great to hear! Thanks, yes please close it. Thanks for testing. As far as
moving to production. This version adds extra complexity so will likely
make this a separate version that can be activated if wanting to use more
than 8 Tilts. I’ll let you know when this goes live.

closed 2020-08-17 by MattL-NZ

MattL-NZ, 2021-08-10:
Hey @noahbaron did you ever manage to push this updated version live? I am planning on setting this up on a completely new Raspberry Pi and just thought I would check the setup is still the same as mentioned above?

noahbaron, 2021-08-10:
Hi @MattL-NZ - yes, this feature is now available in the latest Tilt Pi download. It needs to be enabled in the Tilt Pi  “System” menu to work.

fullbreed06, 2021-09-10:
Hey @noahbaron i just loaded the Tilt Pi v.2 Buster Jan12 image to a PI 4, and i'm having trouble with it picking up all of my tilts. I have all different colors, and the only ones that are being picket up are the ones with repeaters. I have the Pi right next to a SS Brewtech conical, and it won't pick it up without the repeater. Am i missing something? What's the range?

noahbaron, 2021-09-13:
We recommend using a Tilt Pro with stainless steel conicals. Are you using
a standard Tilt or Tilt Pro?

noahbaron, 2021-09-13:
Adding your username tag: @fullbreed06

  We recommend using a Tilt Pro with stainless steel conicals. Are you
using a standard Tilt or Tilt Pro?

fullbreed06, 2021-09-13:
Standard tilt. I'm within 5 feet of the tilt pi. The others is maybe 10
feet away ina glass carboy, and is not being picked up.

noahbaron, 2021-09-13:
Glass carboy shouldn't be a problem at all. The range is 30 feet minimum
and up to 100 feet in a glass carboy. You're saying the repeater connects
10 feet away but not the Tilt in the glass carboy? Does the Tilt 2 app
connect to the Tilts in a carboy?

fullbreed06, 2021-09-14:
When you program the tilt PI and run the set up for wifi and which color tilt, I did one for orange and one for black. The tilt pi is only picking up the one that is connected to the repeater. The tilt Pi is right above the SS conical, and picks it up with the repeater, but doesn't pick up the Black one in the glass carboy 10 ft away. The tilt 2 app does pick up both no problem.

noahbaron, 2021-09-14:
It's possible there was some issue starting a log for your black Tilt. Are
you able to connect to the Tilt Pi dashboard? You can connect to it from a
phone or laptop on the same wifi network through the link
http://tiltpi.local:1880/ui or http://*<IP address for tiltpi>*:1880/ui