baronbrew/TILTpi #27: Question: Original Tilt Precision

SillyHats, 2021-03-18:
Looking at the tiltpi interface and the tilt android app I see 1.xxx displayed. When I look at a google sheet produced by the tiltpi it creates data points with more decimal places as shown below. How is the tilt pi calculating with more precision specifically when logging to the google sheet? This is with a tilt regular (with two screw caps) from 2018.

SG
1.00616666666666
1.00616666666666
1.00633333333333
1.00633333333333
1.0060
1.0060
1.00616666666666
1.00616666666666
1.0060
1.00616666666666
1.0060
1.0060
1.00616666666666
1.00666666666666

wiltdavi, 2021-03-18:
I have not reviewed the code to confirm this but I believe that the value you see on the tilitpi interface and tilt android app is set round the value to a specific number of digits.  The Google sheet is using the same data from the tilt as the other items you referenced so this is the only way it makes sense and is common practice when making an interface.

noahbaron, 2021-03-19:
The extra decimal places are a result of interpolation calculations
occurring when at least two calibration points are in use. The Android app
rounds to 4 decimal places while the Tilt Pi does not round at all. This
wasn’t intentional for Tilt Pi, it just hasn’t been updated like the
Android app.

closed 2021-03-19 by SillyHats