baronbrew/TILTpi #30: Ancient version of node-red?

fredsparkle, 2021-06-18:
Was looking to merge some custom flows and GUI's with the tilt flows and noticed that the version of Node-Red was way back in time. Are you stuck on this version because of a bug or security fix?

If a bug it's been my experience that they get fixed in short order after reporting them.

noahbaron, 2021-06-18:
Yes, there was a bug that took over 2 years to fix discussed here:
https://github.com/node-red/node-red/issues/2017. It prevented using http
POST with Google Scripts/Sheets which uses a redirect.

It appears it was fixed in April of this year, so I'll take a look at
upgrading node-red to the latest version. What kinds of updates were you
planning to add?

fredsparkle, 2021-06-22:
IOT to control his brew rig and Alex voice interface, warnings through the house.

noahbaron, 2021-06-23:
Ok, thanks for letting me know. I’ll take a look at the latest node-red
version and see if I can get it to work.