baronbrew/TILTpi #31: Ability to run multiple tiltpi’s on one network

rampantoctopus, 2021-09-05:
Although I assigned each pi a distinct hostname, when trying to run two instances of tilipi, chromium reports a conflict with another local system running the same service. My two fermentation chambers are too far from each other for one pi to do the job— is there a way to set up a second pi on the same network?

noahbaron, 2021-09-20:
Sorry didn't see this earlier. When assigning a new host name to a Raspberry Pi running Tilt Pi you also have to update the "flows" file in the folder /home/pi/.node-red This is because Tilt Pi runs on node-red and node-red requires this. The flows file needs to bec changed from "flows_tiltpi.json" to "flows_tiltpi-2.json" where "tiltpi-2" is your new hostname. I advise changing host names for both Raspberry Pi's to avoid any caching issues from the router. There shouldn't be any conflict with this when accessing either Tilt Pi with a laptop or phone on the same WiFi network.

rampantoctopus, 2021-09-20:
No problem— actually a Facebook user stepped up and gave me the deets on how to change the flows as you described… I’ve got everything working, thanks! Love your products!! I’m four tilts in and I can’t imagine brewing without them. 

I even got extra tubes and gaskets to use in sours. Love them. 

Best wishes,

Mike

noahbaron, 2021-09-22:
Great to hear! Thanks for your feedback and support. Glad to hear the Tilt
Facebook group could answer your question!