baronbrew/TILTpi #32: Cloud logging broken: undefined flow.get('cloudURL')

memcpy-rand-rand-rand, 2021-09-06:
Trying to figure out why my tiltpi is no longer logging to Brewfather.  The toast pops up saying logging request sent, but there is no subsequent toast or information and the request is never received by Brewfather.

* tiltpi can reach the internet.
* brewfather endpoint working correctly (tested using tilt app on phone)
* have had a couple of prior brews work fine on the system
* tiltpi rebooted a couple of times

Dived in to attempt a little Node Red debugging (using node.warn)

* In Setup Cloud Post node, the flow.get('cloudURL') was returning undefined.
* Re-setting the Cloud URL had no effect
* In trying to debug further, I added a warn to the Custom CloudURL node to tell me the value of msg.payload.  Not actually the thing I was trying to look at, so I had to flip the switch "Use Default Cloud URL for All" to show anything.  "msg.payload" was initially "true" when entering the function; which I guess matches the "on" value for the setting.  set it back to off, "msg.payload" back to "false" as expected.  

After that it looks like the flow.get('cloudURL') starts to succeed.

It looks like there's a logic bug buried in the node graph/code which causes this value to be undefined somehow and cause subsequent logging failures.

Happy to provide any logs etc.

noahbaron, 2021-09-08:
Hi,

Thanks for letting me know about this error you're seeing. If there's an
error even after restarting there must something odd that gets saved into
the settings files that are reloaded when the system boots up. To view
these files you can go to http://tiltpi.local:1880/data/global.json or
http://tiltpi.local:1880/data/<color of Tilt in all caps>.json to get per
Tilt settings or global settings that are saved. It would be great if you
could send a copy of one of these. Here's an example of a red Tilt Pro that
I have and accessed the information via
http://tiltpi.local:1880/data/RED.json

{"uuid":"a495bb10c5b14b44b5121370f02d74de","major":78.4,"minor":10639,"tx_power":-59,"rssi":-46,"mac":"d6:0f:1c:ed:38:83","Color":"RED","displayColor":["RED"],"IDbyMAC":false,"Beer":["Test
Data Access",true],"actualSGPoints":"","unCalSGPoints":"","actualTempPoints":"","unCalTempPoints":"","timeStamp":1631059881668,"formatteddate":"9/7/2021,
17:11:21","Timepoint":44446.71622300926,"fermunits":"","SG":1.0639,"hd":true,"Temp":78.4,"tempunits":"°F","doclongurl":"","defaultcloudURL":["https://script.google.com/macros/s/AKfycbwNXh6rEWoULd0vxWxDylG_PJwQwe0dn5hdtSkuC4k3D9AXBSA/exec",true],"logCloudDataCheck":true,"localloggingInterval":15,"loggingInterval":15,"minRSSI":-105,"alphaSG":null,"alphaTemp":null,"numberSamples":null,"smoothSwitch":false}

here's an example of what happens when you turn off default cloud URL and
use a custom URL. I'm thinking that the customcloudURL key value pair
should probably exist regardless of whether a custom cloud URL exists.

{"uuid":"a495bb10c5b14b44b5121370f02d74de","major":78.4,"minor":10638,"tx_power":0,"rssi":-54,"mac":"d6:0f:1c:ed:38:83","Color":"RED","displayColor":["RED"],"IDbyMAC":false,"Beer":["Test
Data Access",true],"actualSGPoints":"","unCalSGPoints":"","actualTempPoints":"","unCalTempPoints":"","timeStamp":1631060019381,"formatteddate":"9/7/2021,
17:13:39","Timepoint":44446.717816909724,"fermunits":"","SG":1.0638,"hd":true,"Temp":78.4,"tempunits":"°F","doclongurl":"","customcloudURL":"http://some_custom_url.com/data","defaultcloudURL":["https://script.google.com/macros/s/AKfycbwNXh6rEWoULd0vxWxDylG_PJwQwe0dn5hdtSkuC4k3D9AXBSA/exec",false],"logCloudDataCheck":true,"localloggingInterval":15,"loggingInterval":15,"minRSSI":-105,"alphaSG":null,"alphaTemp":null,"numberSamples":null,"smoothSwitch":false}

memcpy-rand-rand-rand, 2021-09-08:
Hey Noah,

Thanks for the tips.  I’ll take a look as soon as I get some time.  It’s working fine now, so whatever was in a bad state is fixed.  I’ll see if I can force a repro when this batch is complete.

Fwiw, the pi had been “up” for a few weeks and had reported on another batch using the same tilt colour (I only have one) previously.  The tilt was then removed, so it wouldn’t have got any new data for a few weeks.  I then did another batch, dropped it in the fermenter in range of the pi and it started picking up the bt beacon ok.  But for some reason had lost connection to the internet – even though I could reach it just fine on my LAN.  Took a router and pi reboot to get it talking to the internet again.

That’s when I noticed the cloud reporting was busted.

Thinking about what else I did... I changed the name of the beer.  I rebooted the pi itself a couple of times ahead of rebooting the router as well.  I think I clicked to update the flow iirc.

Cheers,
Will.

http://tiltpi.local:1880/data/RED.json