baronbrew/TILTpi #33: http://*.*.*.*:1880/data/BLACK.json isn't updating.

flyize, 2021-11-01:
I'm hoping to query the JSON to pull data into my Home Assistant. JSON seems like it would be an easy way to do it, but it doesn't seem like the JSON updates very regularly. My current JSON shows a 10/10/21 date even though its still recording data right now.

Is there a better way to get this data?

noahbaron, 2021-11-01:
The method you're using only updates when a setting or calibration has changed. To get the latest data as JSON format you will need to update the Tilt Pi to the latest version by going to System menu and click the "Update App (flow)" button. (Note the version number will stay the same since this is still a new feature). Once you update the app, you can use the following to access the data: 

http://*.*.*.*:1880/macid/all or http://tiltpi.local:1880/macid/all

You can also request a single MAC ID by replacing the word "all" above with the actual MAC ID of the Tilt you want data on. For example:

http://tiltpi.local:1880/macid/e4:a7:1a:88:a5:9b

Let me know if that works for you.

flyize, 2021-11-02:
Hey, that's fancy! Thanks!