baronbrew/TILTpi #35: Tilt Pi App not seeing PI yet aioblescan does

cvanallen99, 2022-11-10:
I realize this is totally not supported but hoping someone has an idea. My raspberry Pi 4B is running gentoo instead of Raspbian. I've managed to get everything installed per the instructions.  aioblescan can see my tilt just fine when I run the scan command. Node-red doesn't seem to have any errors.  The web interface on :1880/ui comes up fine.

Everything looks great EXCEPT that my Pi never shows up in the web app.  OS level I can see it fine with aioblescan but it never shows up in the GUI.  Hoping someone has a suggestion.

noahbaron, 2022-11-10:
I might have the answer for you. Recently an issue was discovered with
Ubuntu Linux where a deprecation warning was causing a continual reset. Do
you see any warnings when you run the python script Aioblescan?

I can update the script so that it ignores any warning.

cvanallen99, 2022-11-10:
Why as a matter of fact, yes.  I do get a DeprecationWarning.  Specifically I get "DeprecationWarning: There is no current event loop"

noahbaron, 2022-11-10:
Ok, thanks for confirming. I can update the script/flow  to ignore this
warning. I’ll let you know when an update is available.

noahbaron, 2023-02-12:
Hopefully this is fixed now with version 2.8

cvanallen99, 2023-02-13:
thank you!  I'll try to pull it down this week to test.

pinpoint90, 2023-04-05:
I have installed Kali Linux on raspberry pi 4b and experiencing the same problem. I managed to get everyting installed without issues with the following commands:

sudo apt-get update
sudo apt-get install python3-distutils
wget https://github.com/baronbrew/aioblescan/archive/master.zip
unzip master.zip
cd aioblescan-master/
python3 -m pip install aioblescan
bash <(curl -sL https://raw.githubusercontent.com/node-red/linux-installers/master/deb/update-nodejs-and-nodered) --nodered-version="0.18.4"
cd ~/.node-red 
sudo -H npm install node-red-dashboard@2.15.5
sudo systemctl enable nodered.service
sudo reboot
wget -O /home/pi/flow.json https://raw.githubusercontent.com/baronbrew/TILTpi/Aioblescan/flow.json
curl -X POST http://localhost:1880/flows -H "Content-Type: application/json" -H "Node-RED-Deployment-Type: nodes" --data "@/home/pi/flow.json"

I am running the pi app and aioblescan on default user account "pi". 
Aioblescan seems to ping the device using the "sudo python3 -u -m aioblescan -T" but it does not show up on the web interface. 
Any ideas?

noahbaron, 2023-04-05:
Strange, I haven’t tested it yet on non Raspberry Pi OS systems but it
should work.

Can you let me know if you see any errors? To do this go to

http://tiltpi.local:1880/

You should see the node-red flow and possibly some errors in the debug
panel.

Regards

pinpoint90, 2023-04-05:
Ok!
This is from debug:

0.18.4 Node-RED[ Deploy](http://192.168.50.173:1880/#) 
  
5.4.2023, 18:26:57e95c8a86.0f73c8
msg : error
"SyntaxError: Unexpected token u in JSON at position 0"
5.4.2023, 18:27:01node: 1d0659ea.6c2c26
msg.payload : string[27]
"python3 -u -m aioblescan -T"
5.4.2023, 18:27:01node: 1d0659ea.6c2c26
msg.payload : string[19]
"pkill -f aioblescan"
5.4.2023, 18:27:01node: 1d0659ea.6c2c26
msg.payload : string[164]
"sudo: a terminal is required to read the password; either use the -S option to read from standard input or configure an askpass helper↵sudo: a password is required↵"
5.4.2023, 18:27:02node: 1d0659ea.6c2c26
msg.payload : string[135]
"sudo: a terminal is required to read the password; either use the -S option to read from standard input or configure an askpass helper↵"
5.4.2023, 18:27:02node: 1d0659ea.6c2c26
msg.payload : string[29]
"sudo: a password is required↵"
5.4.2023, 18:27:04node: 1d0659ea.6c2c26
msg.payload : string[27]
"python3 -u -m aioblescan -T"
5.4.2023, 18:27:04node: 1d0659ea.6c2c26
msg.payload : string[19]
"pkill -f aioblescan"
5.4.2023, 18:27:04node: 1d0659ea.6c2c26
msg.payload : string[164]
"sudo: a terminal is required to read the password; either use the -S option to read from standard input or configure an askpass helper↵sudo: a password is required↵"
5.4.2023, 18:27:04node: 1d0659ea.6c2c26
msg.payload : string[135]
"sudo: a terminal is required to read the password; either use the -S option to read from standard input or configure an askpass helper↵"
5.4.2023, 18:27:04node: 1d0659ea.6c2c26
msg.payload : string[29]
"sudo: a password is required↵"
5.4.2023, 18:27:09node: 1d0659ea.6c2c26
msg.payload : string[27]
"python3 -u -m aioblescan -T"
5.4.2023, 18:27:09node: 1d0659ea.6c2c26
msg.payload : string[19]
"pkill -f aioblescan"
5.4.2023, 18:27:09node: 1d0659ea.6c2c26
msg.payload : string[164]
"sudo: a terminal is required to read the password; either use the -S option to read from standard input or configure an askpass helper↵sudo: a password is required↵"
5.4.2023, 18:27:09node: 1d0659ea.6c2c26
msg.payload : string[164]
"sudo: a terminal is required to read the password; either use the -S option to read from standard input or configure an askpass helper↵sudo: a password is required↵"
5.4.2023, 18:27:12node: e95c8a86.0f73c8
msg : error
"SyntaxError: Unexpected token u in JSON at position 0"
5.4.2023, 18:27:14node: 1d0659ea.6c2c26
msg.payload : string[27]
"python3 -u -m aioblescan -T"
5.4.2023, 18:27:15node: 1d0659ea.6c2c26
msg.payload : string[19]
"pkill -f aioblescan"
5.4.2023, 18:27:15node: 1d0659ea.6c2c26
msg.payload : string[164]
"sudo: a terminal is required to read the password; either use the -S option to read from standard input or configure an askpass helper↵sudo: a password is required↵"
5.4.2023, 18:27:15node: 1d0659ea.6c2c26
msg.payload : string[164]
"sudo: a terminal is required to read the password; either use the -S option to read from standard input or configure an askpass helper↵sudo: a password is required↵"
5.4.2023, 18:27:19node: 1d0659ea.6c2c26
msg.payload : string[27]
"python3 -u -m aioblescan -T"
5.4.2023, 18:27:19node: 1d0659ea.6c2c26
msg.payload : string[19]
"pkill -f aioblescan"
5.4.2023, 18:27:19node: 1d0659ea.6c2c26
msg.payload : string[164]
"sudo: a terminal is required to read the password; either use the -S option to read from standard input or configure an askpass helper↵sudo: a password is required↵"
5.4.2023, 18:27:19node: 1d0659ea.6c2c26
msg.payload : string[164]
"sudo: a terminal is required to read the password; either use the -S option to read from standard input or configure an askpass helper↵sudo: a password is required↵"
5.4.2023, 18:27:24node: 1d0659ea.6c2c26
msg.payload : string[27]
"python3 -u -m aioblescan -T"
5.4.2023, 18:27:24node: 1d0659ea.6c2c26
msg.payload : string[164]
"sudo: a terminal is required to read the password; either use the -S option to read from standard input or configure an askpass helper↵sudo: a password is required↵"
5.4.2023, 18:27:43node: e95c8a86.0f73c8
msg : error
"SyntaxError: Unexpected token u in JSON at position 0"
5.4.2023, 18:27:44node: 1d0659ea.6c2c26
msg.payload : string[19]
"pkill -f aioblescan"
5.4.2023, 18:27:44node: 1d0659ea.6c2c26
msg.payload : string[164]
"sudo: a terminal is required to read the password; either use the -S option to read from standard input or configure an askpass helper↵sudo: a password is required↵"
5.4.2023, 18:27:44node: 1d0659ea.6c2c26
msg.payload : string[27]
"python3 -u -m aioblescan -T"
5.4.2023, 18:27:44node: 1d0659ea.6c2c26
msg.payload : string[164]
"sudo: a terminal is required to read the password; either use the -S option to read from standard input or configure an askpass helper↵sudo: a password is required↵"
5.4.2023, 18:27:44node: 1d0659ea.6c2c26
msg.payload : string[19]
"pkill -f aioblescan"
5.4.2023, 18:27:44node: 1d0659ea.6c2c26
msg.payload : string[164]
"sudo: a terminal is required to read the password; either use the -S option to read from standard input or configure an askpass helper↵sudo: a password is required↵"
5.4.2023, 18:27:44node: e95c8a86.0f73c8
msg : error
"SyntaxError: Unexpected token u in JSON at position 0"

noahbaron, 2023-04-07:
Thanks! It appears Kali Linux requires the sudo password to start the
Bluetooth scanner. You can disable this feature on Kali Linux by
following the instructions here. Apparently Raspberry Pi OS has this
disabled by default.

***@***.***/how-to-stop-that-annoying-sudo-password-prompt-in-linux-b2b72b9c2f55

pinpoint90, 2023-04-07:
Thank you! the sudo errors are now gone. The only error left is "Unexpected token u in JSON at position 0". I will ty to check if it works with Tilt when I get home.

noahbaron, 2023-04-07:
Sounds like it should work now. The other error won’t prevent the scanning
to start.

pinpoint90, 2023-04-08:
Tested today and it works. Thanks again! :)

noahbaron, 2023-04-09:
Great to hear! Thanks for confirming.

closed 2023-04-11 by noahbaron